MediaTek-Labs/BlocklyDuino-for-LinkIt #22: MCSLite Block 無法加 Channel

bearwbearw, 2017-05-24:
[image]

缺少下面可以加 Channel 的缺口

r04922159, 2017-05-24:
感覺 MCSLite 與 7697 溝通的方式，跟 MCS 與 7697溝通的方式不一樣，MCSLite 應該沒有自己的 Library，跟板子溝通時也沒有 mcs.addChannel() 這種動作，所以在做 Block 的才沒有做缺口給他。

pablosun, 2017-05-25:
`MCSLiteDevice` 跟 `MCSDevice` 除了 constructor 不一樣之外，其他完全一樣，支援相同的操作。

在實做上，`class MCSLiteDevice : public MCSDevice` 是繼承關係。

bearwbearw, 2017-05-25:
如 Pablo 描述, MCSLite 的 block, 除了需要多帶 IP, Port, 其他行為應該是一致的.
請幫忙補上下面那可以新增 Channel 的接點

x43x61x69, 2017-05-25:
c.c. @r04922159

r04922159, 2017-05-25:
好的，我本來看到的是公司這份範例code ( https://github.com/Mediatek-Cloud/mcs-lite-example/blob/master/Arduino/LED_controller/LED_controller.ino )，所以以為 MCSLite 只會透過 RESTful API 跟 Websocket 來溝通。

r04922159, 2017-05-25:
想確定一下，所以現在是希望 MCSLite 跟 MCS 所有行為都共用一個 Block ? 這好像是之前沒有提過的要求，我覺得改動性蠻大的，所以得跟您討論一下。
我覺得並不是動作相同就可以共用，因為 我們之前的 Block 並沒有設計成告知它說是 MCS 還是 MCSLite，所以 generator 生出來的 code 都是 for MCSDevice。

如果要把 MCSLite 加進來 ( 並且可用 )，我能想到兩個能做的方法：
把 MCSLite 獨立出去成一個新的項目，裡面的 Blocks 跟 MCS 的一樣，但是全部都是 for MCSLite。
或
把現存的 MCS 裡面所有的 Block 都加一個下拉選單，可以選擇 MCS or MCSLite，generator 再依此為據決定生成哪個 Device 的 code。

我自己覺得第一項比較 OK，因為第二項加一個下拉選單的話有些 Blocks 我會覺得有點亂 ( 例如新增頻道的Block，又要下拉選單選 Datatype，又要下拉選單選 Device，覺得稍微麻煩一些，而且每個 Block 都要選，操作上也會變得冗長 ) 。

大概想的到這兩種做法，想討論下，不知道您怎麼說 ? 覺得該採取哪個做法，確定之後我再下手，或是您有想到其他可行的做法，可以提出來然後討論討論能否達成。

pablosun, 2017-05-25:
Hi @r04922159 , `MCSLiteDevice` 就是 `MCSDevice`, 所有method都是共用的...

同樣一個Arduino sketch，MCS跟 MCSLite 的差別就只有
```cpp
//MCSDevice mcs("DS4TArHi", "zXfcfyGwejw6Wr1i");
MCSLiteDevice mcs("DS4TArHi", "zXfcfyGwejw6Wr1i", "192.168.1.1", 3000);
```
這樣的差別而已，其他都是完全一模一樣的，這應該只需要添加一個會生成 `MCSLiteDevice` 的 block 就可以了？

r04922159, 2017-05-25:
@pablosun ，我原先設想是說 MCSLite 跟 MCS 放在一個項目下，那可能會有同時使用的情況。這樣子的話因為變數名重複宣告，一份 code 裡 MCS 跟 MCSLite 只能同時使用一個? 但是這個做法是可以正確編譯的，那我就照這樣子改。

pablosun, 2017-05-25:
@bearwbearw @x43x61x69  請 Comment 是否要允許同一個 Blockly Duino Project 可以同時使用 MCS + MCSLite?

@r04922159  變數的名字我以為是可以動態決定的?  還是必須要先寫死呢? 比如 BLE Attribute 可以加很多個，會自動gen出不同的變數名稱，是怎麼做到的呢?

r04922159, 2017-05-25:
@pablosun 目前是mcs那邊寫死，但是是可以用下拉選單變成動態決定，寫死是因為我當時寫的時候不知道有 MCSLite 這東西。

如果要改成動態決定，那就是我前面有提到的用下拉選單先選要 MCS 或是 MCSLite，然後再以此決定變數宣告名稱。然後剩下的 Blocks 也要有個下拉選單決定說這次這個 block 是要 MCS 還是 MCSLite 的動作。

pablosun, 2017-05-25:
瞭解～ 請 @bearwbearw 裁示一下是要加上 dropbox，還是要預設用同樣的名字，不允許 MCS+MCSLite 同時使用？

x43x61x69, 2017-05-25:
感覺可以選是不錯的方式，不過一般來說一張板子要同時使用 MCS 和 MCSLite 會有點奇怪？通常使用其中一個就是因為另一個相對來說比較不和需求（雲端 vs 本機），加上我們對其他功能方面簡化的傾向，目前強制限制只能用一個我覺得不會不合理。

bearwbearw, 2017-05-25:
只會用 Mcs or mcslite, 二選一

pablosun, 2017-05-25:
@r04922159 這樣應該可以確定是 [一份 code 裡 MCS 跟 MCSLite 只能同時使用一個] 的作法了~

r04922159, 2017-05-25:
已修正

closed 2017-05-25 by r04922159