MediaTek-Labs/BlocklyDuino-for-LinkIt #34: BLE Beacon : 新增一個 EddyStoneURL 的 block

bearwbearw, 2017-05-25:
[image]

Gen 出的 code 參考~
// configure our advertisement data as iBeacon.
LBLEAdvertisementData beaconData;

// config URL
beaconData.configAsEddystoneURL(EDDY_HTTPS, “ goo.gl/EHnlj5”);

// start advertising it
LBLEPeripheral.advertiseAsBeacon(beaconData);

可以參考 Arduino 的 Example :

[image]
#

x43x61x69, 2017-05-25:
註：不確定能否限制字數，但是我們可以產生 code 時截掉過長的部分。字數限制可能要用文字提示 or tooltip 的形式。

@r04922159

r04922159, 2017-05-25:
已新增，目前字數限制是用文字提示，可再討論

r04922159, 2017-05-25:
補充：如果URL超過17會自動cut down到前17個

bearwbearw, 2017-05-26:
幫忙研究一下是否有辦法只輸入到 17 的字就無法輸入?
輸入一堆, 在 gen code 階段 cut down 到 17 個這行為太隱晦.

若真的不行, 請和我說一下, 可能在介面上要做些提示~

closed 2017-06-02 by x43x61x69