MediaTek-Labs/BlocklyDuino-for-LinkIt #52: 英文 Digital

bearwbearw, 2017-05-31:
[image]

x43x61x69, 2017-06-02:
Ref. #70.

bearwbearw, 2017-06-02:
請依照上圖修改, 怕漏掉有問題可以當面在旁邊一起改

x43x61x69, 2017-06-02:
`dev` 內都已經是改過的樣子，除非有拼字錯誤否則不會再更動。

許多元件當初是由多個作者接力完成，而我們自行開發的模組並不是採用小寫原則，大規模的修改對使用的幫助有限，**且英文介面目前無使用者**，基於時間成本考量，經討論後決議是視情況和必要性**做最小非功能性更動**。

因為有很多這類大小寫的修改，我們這邊統一規則成：

- **一律大寫**：欄位名稱非敘述句者，採用各字首大寫（例：`Wi-Fi Password`），敘述句則僅句首大寫（例：`Create array with`）。
- **一律小寫**：上述規定外，若該英文字為原始 Arduino 函數名稱（例：`delay()`、`setup()`）。
- **函數名稱**：中文版採用對應意思翻譯函數名稱（`setup()` -> 初始化），英文採用原始 Arduino 函數名稱不更動（`millis()`、`delay()`，視 block 配置後綴括號組`()`），因使用者可以直接參照 Ardunio 官網原始英文解說，且英文不對中文做反向對齊。

@alieonsido @r04922159

nalanala, 2017-06-03:
考慮到修改成本，可能優先修改有必要的項目。

bearwbearw, 2017-06-04:
我們的使用者不是 Arduino user. 不能須從 arduino mapping 回 blocks.
思維應該是從 blocks 要能自己就明確說明清楚.
安排一個會議討論一下吧.

closed 2017-06-09 by bearwbearw