MediaTek-Labs/BlocklyDuino-for-LinkIt #79: Grove 支持

bearwbearw, 2017-06-13:
請幫忙實作四個 Grove Sensor (在 starter kit 內的)
- Ultrasonic Ranger
- T&H Sensor
- LED Bar
- OLED Display

其他 Alpha 版本內未測過的功能, 正式 release 都先關掉
詳細 spec 用郵件寄出, 有任何不清楚地請找我討論.

x43x61x69, 2017-06-19:
因人力及時間不足，Milestone 2.8.0 延後至 6/30 日發佈。

x43x61x69, 2017-06-20:
關於 Grove 元件：

- 官方 Wiki 使用 Port，統一使用 Grove Port / Grove 連接阜。
- Port 原先是針對 Grove Breakout 設計（D2-8, A0-4），加入 D10-12 但不移除既有的。
- Port 下拉選單架構無法做預設選項設定，且預設選項亦無法阻止使用者選擇其他選項，故不做特殊處裡。
- `define` 為無必要性時，考量多個元件並存不在額外定義，直接使用連接阜避免產生錯誤。

註：以上內容將視情況增補。

bearwbearw, 2017-06-23:
同意 Grove 的規則
2~12 麻煩請改成 D2~D12 (要加上 D).
因為這是描述 Grove Port, 不是 Pin.
Grove 的 port 名稱是有包含 D 的.

[image]

x43x61x69, 2017-06-29:
See #128.

bearwbearw, 2017-07-03:
Zw 我這 issue 沒有更進一步問題了. 若你沒問後續問題就關掉吧~

closed 2017-07-03 by x43x61x69