MediaTek-Labs/BlocklyDuino-for-LinkIt #84: Blocks 沒放在 "初始化" 和 "重複執行", 就不會去產生 code

bearwbearw, 2017-06-15:
今天參加教師研習營, 多位老師分別提出這樣的建議.
因為 Block 編輯時有時要 debug, 或是暫時不用, 會把那些 block 拖到旁邊.
但目前還是會產生 code, 所以只好把 Blocks 都刪除, 很不方便.

老師希望 blocks 不再 "初始化" 和 "重複執行" 兩個 blocks 內的都會直接 ignore 掉

x43x61x69, 2017-06-15:
一些關於這個工具的出發點和設計原則：

1. 工具的初衷都是希望使用者可以了解正確的程式設計觀念，希望有新的學習者最後可以去撰寫真正的 code，因為圖形化 IDE 有其支援極限，目的在入門前提供更簡易的對照管道。
2. Blockly 的設計概念本來是於理想的狀況下是可以在任意位置產生正確的 code 結構，沒有向前檢查的概念，要改就要每一個關聯 block 單獨修改（目前也不確定是否真的能這樣改）。
3. 當初新增 setup()/loop() block 是為了要讓使用者對照 Arduino IDE 內的形式方便前述概念的傳達，所以部分抵觸第二點的原則，但是立意是良善的，當初說好的前提是「放在這兩個區塊外我們不支援」，現在使用者違反預先規定的原則，不在支援範圍內。

是故，這項要求一來不符合原先 Arduino IDE 使用原則（你不能合理化寫一段沒用的 code 只因為「省下選起來按一次 Delete 鍵」），也不符合程式設計觀念，我認為老師不能為了教學方便最後培養出來的學生其實根本沒有能力延伸學習或判斷什麼是正確的，方便應該建立在觀念正確之上（了解為什麼可以跟不行），保持 code base 乾淨也是一項好習慣，每一個顯示在畫面上的東西都應該是有意義的。

closed 2017-06-15 by x43x61x69

pablosun, 2017-06-15:
使用者的需求是確實存在的：在開發程式的過程當中，開發者需要一個「**可以暫時存放一些程式想法**」的地方。

舉例來說，我們可能會開一個新檔案暫時寫一些 function 或 code snippet，最後要用的時候再加入project中。

而目前的 Blockly editor 並沒有這樣的一個場所。從「積木」的類比來延伸，我們在製作積木的時候，也時常會先組好一些元件，放在旁邊，等到要用的時候，再調整一下，置放入最後的成品當中。

如果在設計哲學上，我們覺得 [Canvas] 是對應到「程式」，而不是「工作區」的概念的話，那問題就會變成能否提供「工作區」的元件？

 * 比如說，多一個「暫存」的 panel?

@bearwbearw 跟 @x43x61x69 覺得如何？

bearwbearw, 2017-06-15:
我個人認為, 我們明確定義了在 "初始化" "重複執行" 內的 code 才是正式有意義的 code. 在外面其他畫布上的 code 其實就是 "未定義行為". 在使用者明確的有**佔存區**的需求, 用這未定義的畫布區來做佔存區, 可能是個不錯的做法.

老師也明確反應, 學生之前都學過 scratch, 這種放在外面零散的 block 不會加到正式的程式內, 是scartch 的標準做法, 學生也對這種方式很熟悉.

alieonsido, 2017-06-15:
(探頭

我個人覺得 + 自己也有蠻多的親身教學經驗 + 自己也有開過課授課的經驗，
弄清楚每一個教學用的 IDE 的核心思想是一件很重要的事情。

的確，Scratch的整個workpanel可以放一堆Block，
但那是立基在以下兩點的大前提：
1. 支援很半吊子的 parallel 寫法
2. 軟體本身只要求學員能夠讓東西動起來，而不是Workflow.

但討論回Blocklyduino，到目前為止的核心思想應該是屬於幫助初階使用者銜接 Arduino IDE而用，也因此更加地強調Workflow的意義，
也不再會有像是 S4A 那種為了亮五顆LED，卻拉五個點亮LED的Block然後各自被Flag觸發的那種有些不合邏輯的Code。

不需要可以任意擺放Block的 Workpanel 的一個很重要的原因就是為了要強調 Code 是要乾淨的，
不需要的Code不應該存在於你的工作區。
這是一款訓練使用者成為一個能夠正常撰寫Arduino Code的軟體，而不是一個開發軟體，
也不是像Scratch那樣只要求程式設計的「體驗」，我覺得這點也是Blocklyduino難能可貴的地方。

x43x61x69, 2017-06-15:
我做了一個比較折衷，改動上也比較不用大規模的做法，採用 Pablo 所謂暫存區的概念在程式流程中加入一個新的 block「暫存區」，目前已經 merge 進 dev。（見：#86）

這個區域內的所有 blocks 都不會被解析，如果使用者有需要可以把把東西放進這個區塊之後取用。

bearwbearw, 2017-06-15:
若是這樣，就他們拖到註解掉的方塊裡就好了～

x43x61x69, 2017-06-15:
如果有用過註解 block 的話，應該已經注意到註解 block 一樣會被解析，而且不會註解掉全域 setup() 的設定，兩個用途邏輯是也是不同的。