MediaTek-Labs/BlocklyDuino-for-LinkIt #85: loop() 內, 第二行之後產生的 code, 會多縮排一個 tab

bearwbearw, 2017-06-15:
[image]

x43x61x69, 2017-06-15:
#87（註：正確來說是第一行少兩個空白。）

bearwbearw, 2017-06-15:
Arduino 標準的是兩個空白

x43x61x69, 2017-06-15:
Auto indent 是會隨 IDE 設定改變的，不是定值。

bearwbearw, 2017-06-15:
Arduino 的 default 應該是 2 spacing?

bearwbearw, 2017-06-15:
Arduino IDE's default is 2 spacing?

bearwbearw, 2017-06-16:
剛去 Arduino IDE 找, 找不到改indent 的地方.
網路上爬文是要取另外改一個 preference 檔案啥的.

若 IDE 設計不是讓人好改, 我們就應該假設 Arduino IDE 的 tab indent = 2 spacing, 請以這方向來修改 gen 出來的 code.

x43x61x69, 2017-06-16:
當使用者使用 external editor 時，是無法做編輯的，他預設的縮排其實也對這裡沒有影響，再者假使他真的需要編輯，只要整個剪下再重新貼上就會變成他 IDE 設定的格式。縮排在 .conf 裡面，有人會去改，這是習慣問題不管統一哪一種都會有人不一樣，這裡縮排是閱讀上的便利，空白會被 compiler 忽略不影響功能。

closed 2017-06-16 by x43x61x69

bearwbearw, 2017-06-16:
我們的 TA (中小學生) 是不會修改的, 應該用是 default 值會比較好.

x43x61x69, 2017-06-16:
Blockly 原生支援不只一種語言，多數語言建議是 4 spaces，考量相容性當初設計預設就是 4 spaces（例如 Python 官方已經考慮移除 2 spaces 支援，Python 沒有 ; 結尾的概念，全部用 indent）。

這裡顯示上已經統一，我覺得沒有必要不斷爭論這類的非功能細節，使用者理解完全沒有影響，教學者甚至可以透過這個機會解釋為何空白和縮排沒有影響。

bearwbearw, 2017-06-16:
可是 Arduino 原生就 2 spaces 呀....我們這專門就是支持 Arduino 的, 不是 Python. (雖然很多人幹譙 Arduino 的 2 space 不符合主流, 但他畢竟就是存在的事實).
對於在 BlocklyDuino 產生 code, 再到 Arduino 後續編輯修改的人, 4 space 就是和 Arduino default 的 2 space 不同, 反而會早成使用上的困擾.