MediaTek-Labs/BlocklyDuino-for-LinkIt #116: 繼電器

bearwbearw, 2017-06-22:
(1) 請改成 D 的
[image]

[image]

(Note : 拿掉 D2, 因為不知為何初始電位是高電位, 會造成一開機就先通電)

(2) 請改用用 狀態 開/關, 不要用高低電位.
[image]

(3) 圖片請換成
[image]

(4) 埠

x43x61x69, 2017-06-22:
- 2 程式碼與官方 Wiki 說明皆無 On/Off 描述，而是 HIGH/LOW，用與數位 I/O 狀態一致。
- 順帶一提，老師回饋結果顯示老師傾向使用與原始代碼相同的描述，且希望在簡單模組之外有其他更細的模組可以使用，老師希望此工具並不僅限教學，希望可以用來做日常程式編寫的簡單化操作。
- 3 請說明現有圖片有誤之處，如有誤植會再行開啟此 Issue。

bearwbearw, 2017-06-23:
3. 統一換成 Seeed 官方的圖片 (斜視圖)

closed 2017-07-03 by bearwbearw