MediaTek-Labs/BlocklyDuino-for-LinkIt #127: 創建新積木於"數學"種類下無法成功生成

Partical777, 2017-06-28:
創建一個math_degrees_to_radian於math.js底下

1.  https://github.com/BlocklyDuino/BlocklyDuino/wiki/CreateNewBlock 
2.  https://hackmd.io/s/Hy9ivGMpx

已遵循1.wiki之步驟和新增了2. 的所需文字

[image]
[image]

然後於index.html , category.xml 及 main.html ( offline-edition ) 都新增了<block></block>

[image]
 
但是只要index.html新增了<block type="math_degrees_to_radian"></block>
按下"數學"後積木會全部重疊且無法拖曳

[image]

x43x61x69, 2017-07-03:
@Partical777 新增至選單並不是由 `index.html`，而是 `category.xml`，之後透過 `make.sh` 去編譯後自動生成 `index.html`。

你也可以提供以上的程式碼（文字，非截圖）和對應的字串值，我們可以幫你增加至 BlocklyDuino 中。

alieonsido, 2017-07-03:
您好：
原則上應該是不用動到index.html以及main.html的
當您在blockly資料夾的子資料夾改動完以後，在主資料夾  (BlocklyDuino)  運行 'sh make.sh' 即可。

alieonsido, 2017-07-04:
先行暫時Close這個Issue，並對說明文件的內文做重新編排，
如果有問題歡迎再提出！

nalanala, 2017-07-04:
最後要sh make.sh 。但是記得要解決換行字元問題(windows底下要換行格式unix)

x43x61x69, 2017-07-04:
@alieonsido Reopen，因為非 Collaborator 會無法回應 Closed 的 Issue。

x43x61x69, 2017-07-05:
@Partical777 目前最新版本的 dev branch 已經加入角度轉弧度的 block（#143），歡迎參考相關的程式碼。

closed 2017-07-05 by x43x61x69

Partical777, 2017-07-06:
了解，謝謝你們的解惑