MediaTek-Labs/BlocklyDuino-for-LinkIt #129: 以下範例產生的 code 有錯誤

bearwbearw, 2017-06-29:
請 import 這 XML
(使用的是 playground 這 branch)

<xml xmlns="http://www.w3.org/1999/xhtml"><block type="initializes_setup" id="1" x="-37" y="-12"><statement name="CONTENT"><block type="variables_set" id="267" inline="true"><field name="TYPE">Boolean</field><field name="VAR">switch</field><value name="VALUE"><block type="logic_boolean" id="273"><field name="BOOL">FALSE</field></block></value></block></statement><next><block type="initializes_loop" id="2"><statement name="CONTENT"><block type="controls_if" id="264" inline="false"><mutation else="1"></mutation><value name="IF0"><block type="variables_get" id="283"><field name="TYPE">Boolean</field><field name="VAR">switch</field></block></value><statement name="DO0"><block type="grove_relay" id="251"><field name="PIN">3</field><field name="STAT">HIGH</field><next><block type="variables_set" id="288" inline="true"><field name="TYPE">Boolean</field><field name="VAR">switch</field><value name="VALUE"><block type="logic_boolean" id="298"><field name="BOOL">FALSE</field></block></value></block></next></block></statement><statement name="ELSE"><block type="grove_relay" id="307"><field name="PIN">3</field><field name="STAT">LOW</field><next><block type="variables_set" id="312" inline="true"><field name="TYPE">Boolean</field><field name="VAR">switch</field><value name="VALUE"><block type="logic_boolean" id="318"><field name="BOOL">TRUE</field></block></value></block></next></block></statement><next><block type="delay_custom" id="356" inline="false"><value name="DELAY_TIME"><block type="math_number" id="357"><field name="NUM">2000</field></block></value></block></next></block></statement></block></next></block></xml>

新增的是 switch, 但程式裡產生的是 switch2

[image]

[image]

x43x61x69, 2017-06-29:
@bearwbearw 請不要使用右側的選單或是開關 issue，會造成開發團隊不必要的混亂與困擾。

x43x61x69, 2017-07-03:
`switch` 為 C 語言保留字，改用 `_switch` 即會正常。

closed 2017-07-03 by x43x61x69