MediaTek-Labs/BlocklyDuino-for-LinkIt #191: 遙控器問題

jackjeanab, 2018-06-23:
依照官網範例
https://docs.labs.mediatek.com/linkit-7697-blocklyduino/e01-led-12883452.html
編譯時發生錯誤
arduino-1.8.5\arduino_debug.exe --upload sketches/tmp/tmp.ino --board LinkIt:linkit_rtos:linkit_7697 --port COM3 --pref build.path=build/tmp/ --verbose-build
Looking for library 'user32'

 Adding paths from jna.library.path: null
 Trying user32.dll
 Found library 'user32' at user32.dll
Looking for library 'shell32'
 Adding paths from jna.library.path: null
 Trying shell32.dll
 Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
 Trying Ole32.dll
 Found library 'Ole32' at Ole32.dll
Loading configuration...
Initializing packages...
WARNING: Error loading hardware folder C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\hardware\arduino
 No valid hardware definitions found in folder arduino.
Preparing boards...
Invalid library found in C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master
Invalid library found in C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master
Invalid library found in C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master
Invalid library found in C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\Light_Sensor_LM358-master
Verifying...
C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\hardware -hardware C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages -tools C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\tools-builder -tools C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages -built-in-libraries C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\libraries -libraries C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\sketchbook\libraries -fqbn=LinkIt:linkit_rtos:linkit_7697 -ide-version=10805 -build-path C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\build\tmp -warnings=none -build-cache C:\Users\Jack\AppData\Local\Temp\arduino_cache_36686 -prefs=build.path=build/tmp/ -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.flash_tool.path=C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\tools\flash_tool\1.4.4 -verbose C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\sketches\tmp\tmp.ino
C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\hardware -hardware C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages -tools C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\tools-builder -tools C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages -built-in-libraries C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\libraries -libraries C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\sketchbook\libraries -fqbn=LinkIt:linkit_rtos:linkit_7697 -ide-version=10805 -build-path C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\build\tmp -warnings=none -build-cache C:\Users\Jack\AppData\Local\Temp\arduino_cache_36686 -prefs=build.path=build/tmp/ -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.flash_tool.path=C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\tools\flash_tool\1.4.4 -verbose C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\sketches\tmp\tmp.ino
Using board 'linkit_7697' from platform in folder: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18
Using core 'arduino' from platform in folder: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18
Build options changed, rebuilding all
Detecting libraries used...
"C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -mlittle-endian -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DPRODUCT_VERSION=7697 -DMTK_BSPEXT_ENABLE -DUSE_HAL_DRIVER -DMTK_NVDM_ENABLE -DMTK_DEBUG_LEVEL_INFO -DMTK_DEBUG_LEVEL_WARNING -DMTK_DEBUG_LEVEL_ERROR -DMTK_LWIP_ENABLE -DMTK_MINISUPP_ENABLE -DMTK_WIFI_API_TEST_CLI_ENABLE -DMTK_WIFI_REPEATER_ENABLE -DMTK_WIFI_WPS_ENABLE -DMBEDTLS_CONFIG_FILE="config-mtk-websocket.h" -DSUPPORT_MBEDTLS -DMTK_MINISUPP_ENABLE -DMTK_WIFI_TGN_VERIFY_ENABLE -gdwarf-2 -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-builtin -fno-strict-aliasing -fno-common -DPCFG_OS=2 -D_REENT_SMALL "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/driver/chip/mt7687/inc/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/driver/chip/mt7687/src/common/include/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/driver/chip/inc/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/driver/CMSIS/Device/MTK/mt7687/Include/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/driver/CMSIS/Include/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/kernel/rtos/FreeRTOS/Source/include/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/kernel/service/inc" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/kernel/rtos/FreeRTOS/Source/portable/GCC/ARM_CM4F/" "-IC:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\system/linkit_7697/src/middleware/third_party/lwip/src/include/" [... cut ...] uch file or directory

 #include <bits/c++config.h>

 ^

 compilation terminated.

Using library LRemote at version 1.0 in folder: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote 
Using library LBLE at version 1.0 in folder: C:\Users\Jack\Downloads\blocklyduino-3.0.202b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LBLE 
exit status 1

pablosun, 2018-06-25:
目前看來跟路徑深度有關，當路徑深度比較深的時候，c++config.h就會找不到。一般來說，g++應該會依照我們指定的hardware architecture，自行替換掉合適的 internal header path。一個解決方法是直接在綠色包的 platform.txt 強制加上 include path`-I{runtime.tools.arm-none-eabi-gcc.path}/arm-none-eabi/include/c++/4.8.3/arm-none-eabi`。

pablosun, 2018-06-25:
使用這一份`platform.txt` 可以解決此問題。
[platform.txt](https://github.com/MediaTek-Labs/BlocklyDuino-for-LinkIt/files/2131898/platform.txt)

pablosun, 2018-06-25:
@jackjeanab 煩請協助確認：
http://download.labs.mediatek.com/resource/blocklyduino/blocklyduino-3.0.202b-win64-ide.zip
應該有修正此問題。

closed 2018-06-27 by pablosun