MediaTek-Labs/BlocklyDuino-for-LinkIt #194: 遙控器積木問題

jackjeanab, 2018-06-27:
使用更新文字標籤積木時  直接填入字串  編譯錯誤
[image]

jackjeanab, 2018-06-27:
[error.zip](https://github.com/MediaTek-Labs/BlocklyDuino-for-LinkIt/files/2142459/error.zip)

jackjeanab, 2018-06-27:
燒錄狀態
Launching Arduino.exe...
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Build options changed, rebuilding all
In file included from C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\sketches\tmp\tmp.ino:9:0:

C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h: In constructor 'LRemoteClass::LRemoteClass()':

 C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h:365:7: warning: 'LRemoteClass::m_canvasColumn' will be initialized after [-Wreorder]

 int m_canvasColumn;

 ^

 C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h:364:7: warning: 'int LRemoteClass::m_canvasRow' [-Wreorder]

 int m_canvasRow;

 ^

 C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h:276:3: warning: when initialized here [-Wreorder]

 LRemoteClass()

 ^

C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h:366:17: warning: 'LRemoteClass::m_orientation' will be initialized after [-Wreorder]

 RCOrientation m_orientation;

 ^

 C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h:362:10: warning: 'String LRemoteClass::m_deviceName' [-Wreorder]

 String m_deviceName;

 ^

 C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LRemote\src/LRemote.h:276:3: warning: when initialized here [-Wreorder]

 LRemoteClass()

 ^

 C:\Users\Jack\Downloads\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\sketches\tmp\tmp.ino: In function 'void loop()':

 tmp:58: error: 'Counting' was not declared in this scope

 label1.updateText(String(Counting));

 ^

exit status 1
Arduino.exe exited with 1 and null

pablosun, 2018-07-09:
The code generation routine `Blockly.Arduino.linkit_lremote_update_textlabel` wrong fully remotes `"` from the generated expression

commit 50a8f29f (2018-07-09): Fix #194 by fixing linkit_lremote_update_textlabel

There is no need to remove '"' symbol from the generated expression.
closed 2018-07-10 by pablosun