MediaTek-Labs/BlocklyDuino-for-LinkIt #196: AP連線編譯錯誤

jackjeanab, 2018-07-01:
[image]

錯誤訊息
Launching Arduino.exe...
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Build options changed, rebuilding all
In file included from c:\users\jack\documents\temp\blocklyduino-3.0.214b-win64-ide\blocklyduino3\arduino-1.8.5\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\stl_tree.h:61:0,

from c:\users\jack\documents\temp\blocklyduino-3.0.214b-win64-ide\blocklyduino3\arduino-1.8.5\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\set:60,

 from C:\Users\Jack\Documents\temp\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LWiFi\src/WiFiServer.h:25,

 from C:\Users\Jack\Documents\temp\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\arduino-1.8.5\portable\packages\LinkIt\hardware\linkit_rtos\0.10.18\libraries\LWiFi\src/LWiFi.h:31,

 from C:\Users\Jack\Documents\temp\blocklyduino-3.0.214b-win64-ide\BlocklyDuino3\sketches\tmp\tmp.ino:9:

c:\users\jack\documents\temp\blocklyduino-3.0.214b-win64-ide\blocklyduino3\arduino-1.8.5\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\stl_algobase.h:65:42: fatal error: bits/stl_iterator_base_types.h: No such file or directory

 #include <bits/stl_iterator_base_types.h>

 ^

 compilation terminated.

exit status 1
Arduino.exe exited with 1 and null

pablosun, 2018-07-09:
Hi @jackjeanab , 能否請您實驗看看，將 BlocklyDuino3 的目錄，從 `C:\Users\Jack\Documents\temp\blocklyduino-3.0.214b-win64-ide\` 搬移到 `C:\` 或是 `D:\`，看看此問題是否仍會發生呢？

因為我去看那個檔案應該是要存在的，所以有可能跟路徑長度限制有關係。

jackjeanab, 2018-07-10:
用上面的方法  把路徑縮短  問題就不會發生

pablosun, 2018-07-11:
@jackjeanab 感謝協助驗證，我們會再繼續釐清怎麼解決路徑深度（也有可能是 command 長度）的問題，此問題先保持 Open。

jackjeanab, 2018-07-11:
OK 我們先維持較短的工作路徑